WiringPi/WiringPi-Node #16: how can i read the pin mode

cokapp-charlie, 2015-02-06:
There is a core function to set the pin mode named pinMode(pin,mode), but how can i get the pin mode(OUT or IN)?

nekuz0r, 2015-02-06:
No, there is no function returning the configured mode of a pin.
Why would you want this kind of function ? You are supposed to know the mode since you are configuring it at your program start.

cokapp-charlie, 2015-02-12:
Because i want create a restful api like [webiopi](https://code.google.com/p/webiopi/).
The python module RPi.GPIO supply a api to read the pin mode:
`
func = GPIO.gpio_function(pin)
`

dirkels, 2015-05-04:
The function getAlt gives the mode of a pin: 0 = input, 1 = output, 4 = alt0, 5 = alt1, 6 = alt2, 7 = alt3, 3 = alt4 & 2 = alt5.

closed 2015-05-05 by nekuz0r