WiringPi/WiringPi-Node #17: Exception: ../src/addon.cc:30:3: error: ‘ThrowException’ is not a member of ‘v8’

daanoz, 2015-02-12:
When installing wiring-pi 2.0.0 I get multiple exceptions, most related to the V8 namespace, is there some dependency missing?

        Making wiring-pi ... child_process: customFds option is deprecated, use stdio instead.
        make: Entering directory '/home/pi/piCtrl/node_modules/wiring-pi/build'
          CXX(target) Release/obj.target/wiringPi/src/addon.o
        ../src/addon.cc: In function ‘void throw_error(const char*, ...)’:
        ../src/addon.cc:30:3: error: ‘ThrowException’ is not a member of ‘v8’
        ../src/addon.cc:30:43: error: ‘New’ is not a member of ‘v8::String’
        wiringPi.target.mk:126: recipe for target 'Release/obj.target/wiringPi/src/addon.o' failed
        make: *** [Release/obj.target/wiringPi/src/addon.o] Error 1
        make: Leaving directory '/home/pi/piCtrl/node_modules/wiring-pi/build'
        gyp ERR! build error
        gyp ERR! stack Error: `make` failed with exit code: 2
        gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
        gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
        gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1067:12)
        gyp ERR! System Linux 3.18.5+
        gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
        gyp ERR! cwd /home/pi/piCtrl/node_modules/wiring-pi
        gyp ERR! node -v v0.12.0
        gyp ERR! node-gyp -v v1.0.2
        gyp ERR! not ok

Tested on both Raspberry B and Raspberry 2 B running raspbian.
Also tested Node v0.12 & v0.11.9

daanoz, 2015-02-12:
I just managed to get it to work on Node v0.10.35...

nekuz0r, 2015-02-12:
The source code is not ready for >0.10.x

daanoz, 2015-02-12:
Aah ok, thanks. I didn't realize that.

nekuz0r, 2015-02-26:
The next release v2.1.0 will be compatible with <=0.12.x

closed 2015-02-26 by nekuz0r

hevik, 2015-03-10:
how do I downgrade my node version so I can use this package?

nekuz0r, 2015-03-10:
you can reinstall node on a lower version or use NVM : https://github.com/creationix/nvm