WiringPi/WiringPi-Node #19: Does not work with Raspberry Pi 2

Achronite, 2015-02-26:
Can you please update the package to include gordon's new libraries for WiringPi?  I believe you need at least version 24 to enable this to work correctly with the new raspberry Pi 2.

nekuz0r, 2015-02-26:
It's in progress, rpi 2 support will be available with v2.1.0

Achronite, 2015-02-26:
Excellent - Thanks for the quick reply.

nekuz0r, 2015-02-26:
As you can see here : https://github.com/nekuz0r/wiringpi/tree/incoming-v2.25
A new version of the custom libWiringPi will be released in parallel of the node wiring-pi v2.1.0 module

closed 2015-02-26 by nekuz0r

nitram509, 2015-07-12:
I've did a little monkey patching and crafted a wiring-pi fork working on Raspberry Pi 2 with the latest wiringPi code.

This may give @nekuz0r a starting point for migration or helps other people (like @Achronite) as an interim solution, until v.2.1.0 is available.

See my fork for details:
https://github.com/nitram509/wiring-pi