WiringPi/WiringPi-Node #21: wiringPiISR is crashing

nekuz0r, 2015-02-27:
wiringPiISR implementation need to be rewrote using uv_async_send instead of uv_queue_work.

stuarttayler, 2015-03-10:
Would you be able to describe the bug? I was having a problem where wiringPiISR would continuously fire the call back and wondering if it's the same issue...

nekuz0r, 2015-03-10:
It crashes on node 0.12.x.
If the ISR is trigger very quickly, it's because of noises.
The ISR do not trigger on state change it triggers on voltage changes.
If the voltage drop a little a (even few millivolts) you get a FALLING event and on the opposite if the voltage increase you get a RISING event.

So you need to filter your signal, a schmitt trigger + some capacitor is a good way to filter the signal.

stuarttayler, 2015-03-10:
Thanks for the reply and explanation.

The issue is strange because I wasn't having this issue before and I didn't have any capacitors etc. It's only happened since I did a 'sudo apt-get update' and 'sudo apt-get upgrade'.

Now, when I run the following code:
```javascript
var wpi = require('wiring-pi');
wpi.setup('wpi');

// // setup
wpi.pinMode(27, wpi.INPUT);
wpi.pullUpDnControl(27, wpi.PUD_UP);

wpi.wiringPiISR(27, wpi.INT_EDGE_FALLING, function(delta) {
  console.log('Pin 7 changed to LOW	 (', delta, ')');
});

setInterval(function() {
	console.log('waiting');
}, 2000);
```
This is the output I get:
```javascript
waiting
waiting
waiting
waiting
waiting
waiting
Pin 27 changed to LOW	 ( 12908576 )
Pin 27 changed to LOW	 ( 979 )
Pin 27 changed to LOW	 ( 26 )
Pin 27 changed to LOW	 ( 15 )
Pin 27 changed to LOW	 ( 15 )
Pin 27 changed to LOW	 ( 17 )
Pin 27 changed to LOW	 ( 14 )
Pin 27 changed to LOW	 ( 13 )
Pin 27 changed to LOW	 ( 13 )
Pin 27 changed to LOW	 ( 14 )
```
It logs 'waiting' until the button is pressed. The output continues then like this, registering the delta as around 14. Before it only ran the call back once or twice.

I have an equivalent test program in python using RPi.GPIO and interrupts and that seems to work as normal, so wondering what could have caused this change...

thanks

nekuz0r, 2015-03-11:
Please fill a new issue, since it is not related to this one.

closed 2015-03-11 by nekuz0r