WiringPi/WiringPi-Node #22: Unable to open GPIO error

stuarttayler, 2015-03-06:
Hi when I start my program, quite often I get this error in the terminal:

wiringPiISR: unable to open /sys/class/gpio/gpio22/value: No such file or directory

Do you know what may cause this? Is it something to be wrong with my code or with wiring-pi?

Thanks

Stu

nekuz0r, 2015-03-09:
Looks like a libwiringpi error not the node module.
It's related to sys mode, on setup all gpio should be exported, looks like something went wrong on that step.
What raspberry pi are you running your program on ?
What node version ?

stuarttayler, 2015-03-09:
Hi,
I'm running a Raspberry Pi B+, node version v0.10.35.
Could it be something to do with the pins not unexporting on shutdown? I have a hardware shutdown button...
Thanks
Stuart

nekuz0r, 2015-03-10:
I got the exact same error while testing the incoming release of wiring-pi on a rpi2.
Trying to figure out what is causing this issue.
After a reboot it worked just fine, really weird.