WiringPi/WiringPi-Node #29: not working with MCP23s17

nemr, 2015-09-01:
Hi , I have been trying to write high to GPIOA0 on the MCP23s17 chip using the wiring-pi npm . The code is runing without errors but the the GPIOA0 pin is not turning high 

Here is the code I used . 

var wpi = require('wiring-pi');

wpi.setup('wpi');
var pin = 64;
wpi.mcp23s17Setup(pin, 0, 0)

wpi.pinMode(pin, wpi.OUTPUT);

setInterval(function() {
wpi.digitalWrite (pin, 1);
 console.log(pin)
}, 500);

I have tried setting the pin value to 123 , 64 and even increment it from 64 to 140 but nothing happens to any gpio's on the MCP23s17 . 

the reset is connected to 5v , clk , mosi , miso  of the chip is connected to SCLK , mosi , miso on the pi receptively I have also tried connecting  mosi  on the chip with miso on the pi but that did do anything as well

nekuz0r, 2015-09-01:
How are wired the address pins of the MCP23S17 (A0:A2) ?

nemr, 2015-09-01:
all are connected to ground  so devid is 0

nekuz0r, 2015-09-01:
did you wired the CS pin to the pi ?

nemr, 2015-09-01:
yes to CE0

nekuz0r, 2015-09-01:
No clue right now, i am using myself this chip in some projects.

nemr, 2015-09-01:
is it possible to provide me with a sample code u used for this ship if my code is correct so I can test it and see if the problem is in the hardware ? 
and did u connect the MISO from the chip to the MISO or to the MOSI on the raspberry pi ?

Thanks for your help

nemr, 2015-09-02:
??

nemr, 2015-09-07:
Any updates on this ?

peychart, 2017-10-11:
why "return 0" in the mcp23017Setup function? Why not "return fd"?