WiringPi/WiringPi-Node #30: V8 fatal error

Disgust, 2015-10-01:
Hi.
While handling interrupt I got the following error:
FATAL ERROR: v8::HandleScope::CreateHandle() Cannot create a handle without a HandleScope

My code is below:
```javascript
var gpio = require('wiring-pi');
gpio.setup('wpi');

//inputs
var signalStartMovie= 7;
//outputs
var signalMovieEnded= 5;

gpio.pinMode(signalMovieEnded, gpio.OUTPUT);
gpio.pinMode(signalStartMovie, gpio.INPUT);
gpio.pullUpDnControl(signalStartMovie, gpio.PUD_UP);

gpio.wiringPiISR(7, gpio.INT_EDGE_FALLING, function(delta) {
    console.log("Interrupt!", delta);
    //gpio.wiringPiISRCancel(signalStartMovie);
});

setInterval(function () {
    gpio.digitalWrite(signalMovieEnded, gpio.HIGH);
    console.log(gpio.digitalRead(signalStartMovie));
    setTimeout(function () {
        gpio.digitalWrite(signalMovieEnded, gpio.LOW);
    }, 500);
}, 1000);
```
What is this? How to solve?

nekuz0r, 2015-10-19:
Which version of nodejs are you using ?

Disgust, 2015-10-19:
```
C:\Windows\System32>node --version
v0.12.4
```

nekuz0r, 2015-10-21:
Is it reproducible with your code ?
Is it happening directly or random ?

xseignard, 2015-11-29:
Hello same error here.

With Raspbian Jessie Lite, node v4.2.1 and 2.1.0 of this module.

Here is a simple code that makes V8 crash:

```
var wpi = require('wiring-pi');
var btnPin = 4;

wpi.setup('wpi');
wpi.pinMode(btnPin, wpi.INPUT);
wpi.pullUpDnControl(btnPin, wpi.PUD_UP);
wpi.wiringPiISR(btnPin, wpi.INT_EDGE_FALLING, function(delta) {
	console.log('btnPin changed to LOW (', delta, ')');
});
```

SteveMacAwesome, 2015-12-23:
I am getting this issue also, it occurs even when using the sample code in the documentation. 

A workaround could be implemented using polling instead of interrupts but I haven't tested this approach yet.

SteveMacAwesome, 2015-12-23:
PS I'm using Raspbian Wheezy, all up to date.

Illizian, 2016-01-01:
:+1: I'm also seeing this issue, with the example code and a simple push button.

```
$ uname -a
Linux alarmpi 4.1.15-1-ARCH #1 Tue Dec 15 18:28:56 MST 2015 armv6l GNU/Linux

$ node --version
v5.3.0
```

nekuz0r, 2016-01-05:
Fixed, closing

closed 2016-01-05 by nekuz0r