WiringPi/WiringPi-Node #31: Cannot install wiring-pi because of libwiringpi cloning failed

hyourinseta, 2015-10-13:
When i tried to install wiring pi using npm, it failed and shown below error
```
# npm install wiring-pi

\
> wiring-pi@2.1.0 install /root/.node-red/node_modules/wiring-pi
> /bin/bash ./install.sh

Cloning libWiringPi ... failed.

================================================================================
FATAL: Cloning libWiringPi failed.
Please check install.log and fix any problems. If you're still stuck,
then please open a new issue then post all the output and as many details as you can to
  https://github.com/eugeneware/wiring-pi/issues
================================================================================

npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! wiring-pi@2.1.0 install: `/bin/bash ./install.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the wiring-pi@2.1.0 install script.
npm ERR! This is most likely a problem with the wiring-pi package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     /bin/bash ./install.sh
npm ERR! You can get their info via:
npm ERR!     npm owner ls wiring-pi
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.1.7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "wiring-pi"
npm ERR! cwd /root/.node-red
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /root/.node-red/npm-debug.log
npm ERR! not ok code 0
```

The error tells me that I have to tell the wiring-pi author. Can you help me?

nekuz0r, 2015-10-19:
Can you post the content of the file `install.log` in `node_modules/wiring-pi` ?

nekuz0r, 2015-10-23:
Cannot reproduce, maybe you had issue with your internet connection

closed 2015-10-23 by nekuz0r

erikkallen, 2016-03-21:
had the same thing turned out git was not installed
```sudo apt-get install git-core``` fixed it

kms77, 2016-08-10:
I had the same problem and now it is solved thanks to you @erikkallen 👍

IgorGanapolsky, 2017-12-18:
@erikkallen Is there a solution for this on a Mac?

erikkallen, 2017-12-18:
I think git should be on your system, you can check this by typing git in your terminal, if it is not available you could try installing the xcode-commandline tools as described here http://railsapps.github.io/xcode-command-line-tools.html if that does not work I wouldn't know

caiusCitiriga, 2018-01-25:
any news for mac?

tillepille, 2018-01-25:
@IgorGanapolsky  and @caiusCitiriga , this is not supposed to work on a Mac since the required hardware isn't there.. Or did i miss anything?

caiusCitiriga, 2018-01-26:
@tillepille ouch, it's how I was supposing.... So, here's what I'm trying to get. I'm developing my core scripts for the raspi on a Mac, using typescript. So, the @types gets successfully installed, the package though is not. Is there a workaround where I can "mock this whole thing?" I wouldn't like to switch lib, or machine. In fact, the same code ported on the raspi, and ran npm i works like a charm (of course)

caiusCitiriga, 2018-01-30:
@tillepille Well, I appreciate your support anyways :) 

# Workaround
If it can be useful to someone, I've found a workaround. It's ugly, not very practical without setting some scripts in the package, but for me, after some configuration, it turned out to be very useful.

So here's the stuff that you need:
+ Virtualbox (i know, it sounds crazy)
+ A Ubuntu server distro (server because it comes without the GUI by default)
+ A Raspberry Pi
+ A local network
+ A folder where you store all of your code on your mac. Eg: ~/Code

Now, the main problem for us mac folks, is that we cannot build wiringPi, but a ubuntu machine can. So the solution:
+ Link togheter the Mac ***Code*** folder with the Ubuntu VM, using the VirtualBox shared folder
+ The ubuntu VM starts a watcher for file changes (I use Typescript to build the dist)
+ On the VM you will **only** install through npm the wiringpi-node package.
+ The next npm install will be performed from mac
+ With a SSH connection through the machines, and a bit of npm scripts, you can automate the whole build and deploy process

# The setup step by step
### [1] Get the VM ready and set
Install your Ubuntu distro on Virtualbox creating a new VM. 
For the configuration I've chosen:
 + 2 cores
 + 2048 MB of RAM
 + 16 MB of VRAM
 + Bridged connection
 + 10GB of virtual hard drive

**Note:** the ***bridged connection*** is the most important part. 
This one allows your VM to be seen on your local network as a physical machine, and shares the same network with your host.

Create your VM, attach to the virtual CDRom unit the ISO of your distro and boot it up.
Follow the step to install the distro as normal.

**Note:** during the install, when the packages selection comes up, remember to check the SSH. Otherwise you'll have to apt-get install it.

Once the installation is done, log into the machine and assign a static IP. 
Mine was: 
+ 192.168.1.12 for the VM
+ 192.168.1.2 for the Mac
+ 192.168.1.10 for the Pi

In the VM terminal type:
```
sudo nano /etc/network/interfaces
```

This will open up the interfaces configuration file. Here, add your network configuration.
Mine looks like this:
```
# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

source /etc/network/interfaces.d/*

# The loopback network interface
auto lo
iface lo inet loopback

# The primary network interface
auto enp0s3
iface enp0s3 inet static
    address 192.168.1.12
    netmask 255.255.255.0
    network 192.168.1.1
    broadcast 192.168.1.255
    gateway 192.168.1.1
    dns-nameservers 8.8.8.8 8.8.4.4
```
[... 187 lines cut ...]

# Login via SSH on the raspi and launch the following commands:
# - Go to the home folder
# - Unzip the previously received zip file into a folder called core
# - Cd into core
# - NPM I
# - When all done, echo a message and close connection
ssh maxine@192.168.1.10 'cd /home/maxine/; unzip -o core.zip -d core; cd core; npm i; echo; echo ========== TRANSFER COMPLETED ==========; exit;';
```

#### wiringpi-node @types
Unfortunately I've found only the @types/wiring-pi. But for what I've learned from the documentation of the library, the methods are the same. 
So what I've did, was install the @types for wiring-pi (the old lib), inspect the declaration file, copy the content and transform it into a interface. I'll use that interface whenever I have to referr to wiringpi-node. If you want more infos about this, feel free to ask :)

That's all! 

Hope it helps someone, and I apoligise for the lenght. If there's something not clear, please ask. If you can improve this somehow, feel free to do it.

Here's a working repository of a robot that I've built using this solution:
https://github.com/hugo-bot-org/core

### Go further with the scripts usage
If you're a cool guy, you do most of your work from the terminal. 
And if you use custom aliases and functions, you should seriously take a look at this:
https://www.npmjs.com/package/shell-profiler