WiringPi/WiringPi-Node #33: Cannot install on node v5.3.0, gcc 4.6.3

haampie, 2015-12-30:
When using `npm install wiring-pi`, the following errors occur:

```
Cloning libWiringPi ... done.
patching file ./wiringpi/devLib/Makefile
Hunk #1 succeeded at 24 with fuzz 1 (offset -2 lines).
Hunk #2 succeeded at 32 with fuzz 2 (offset -2 lines).
patching file ./wiringpi/gpio/Makefile
Making libWiringPi ... done.
Making devLib ...done.
Unistalling gpio utility ... done.
Making gpio utility ... done.
Installing gpio utility ... done.
Making wiring-pi ... make: Map '/root/tmp/node_modules/wiring-pi/build' wordt binnengegaan
  CXX(target) Release/obj.target/wiringPi/src/addon.o
In file included from ../src/addon.h:4:0,
                 from ../src/addon.cc:1:
/root/.node-gyp/5.3.0/include/node/v8.h:336:1: error: expected unqualified-id before ‘using’
/root/.node-gyp/5.3.0/include/node/v8.h:469:1: error: expected unqualified-id before ‘using’
/root/.node-gyp/5.3.0/include/node/v8.h:856:1: error: expected unqualified-id before ‘using’
In file included from ../src/addon.cc:1:0:
../src/addon.h:12:24: error: ‘Handle’ is not a member of ‘v8’
../src/addon.h:12:44: error: expected primary-expression before ‘>’ token
../src/addon.h:12:46: error: ‘val’ was not declared in this scope
../src/addon.h:13:28: error: redefinition of ‘bool node::Buffer::HasInstance’
../src/addon.h:12:12: error: ‘bool node::Buffer::HasInstance’ previously defined here
../src/addon.h:13:24: error: ‘Handle’ is not a member of ‘v8’
../src/addon.h:13:45: error: expected primary-expression before ‘>’ token
../src/addon.h:13:47: error: ‘val’ was not declared in this scope
../src/addon.h:14:18: error: ‘Handle’ is not a member of ‘v8’
../src/addon.h:14:38: error: expected primary-expression before ‘>’ token
../src/addon.h:14:40: error: ‘val’ was not declared in this scope
../src/addon.h:15:22: error: redefinition of ‘char* node::Buffer::Data’
../src/addon.h:14:13: error: ‘char* node::Buffer::Data’ previously defined here
../src/addon.h:15:18: error: ‘Handle’ is not a member of ‘v8’
../src/addon.h:15:39: error: expected primary-expression before ‘>’ token
../src/addon.h:15:41: error: ‘val’ was not declared in this scope
../src/addon.h:16:21: error: ‘Handle’ is not a member of ‘v8’
../src/addon.h:16:41: error: expected primary-expression before ‘>’ token
../src/addon.h:16:43: error: ‘val’ was not declared in this scope
../src/addon.h:17:25: error: redefinition of ‘size_t node::Buffer::Length’
../src/addon.h:16:14: error: ‘size_t node::Buffer::Length’ previously defined here
../src/addon.h:17:21: error: ‘Handle’ is not a member of ‘v8’
../src/addon.h:17:42: error: expected primary-expression before ‘>’ token
../src/addon.h:17:44: error: ‘val’ was not declared in this scope
```

Using the latest node version (5.3.0) and gcc 4.6.3

nekuz0r, 2016-01-05:
Wiring-pi was never tested on node v5

taoyuan, 2016-01-07:
@haampie I'm using `wiring-pi` in v5. It works. I build it in `gcc version 4.9.2 (Raspbian 4.9.2-10)`. May be you can try `gcc 4.9.2`.

OvisMaximus, 2016-01-17:
Me 2: I want to itroduce wiring-pi in another project as a dependency. This project uses travis CI, where I have no control over the complier or node version. Node version there is v5.4.1, gcc is 4.6.3. Everything is fine on my raspi with the same node version and gcc 4.9.2. Any hint, how to fix this?

msimulcik, 2016-02-27:
I had the same problem with node 4.3.1. I updated gcc to 4.9.2 following [this article](https://solarianprogrammer.com/2015/01/13/raspberry-pi-raspbian-install-gcc-compile-cpp-14-programs/) and it builds successfully now.

nekuz0r, 2016-10-06:
I am closing this issue since it's looks like a GCC issue, fixed by upgrading to 4.9.x

closed 2016-10-06 by nekuz0r