WiringPi/WiringPi-Node #35: Installation error: Cloning of libWiringPi failed.

osmoma, 2016-01-05:
I want to install wiring-pi on my Ubuntu 15.10 laptop, but got some error messages.
I have installed Node 4.x from  https://nodejs.org/en/download/package-manager/#debian-and-ubuntu-based-linux-distributions

And have these versions:
$ nodejs --version  
v4.2.4
$ node --version 
v4.2.4
$ npm --version 
2.14.12
$ lsb_release -a
Distributor ID:	Ubuntu Linux 15.10
Release:	15.10  wily
-------------------------------------------
I have attached /home/moma/npm-debug.log
-------------------------------------------
$ sudo npm install wiring-pi
npm WARN package.json moma@1.0.0 No description
npm WARN package.json moma@1.0.0 No repository field.
npm WARN package.json moma@1.0.0 No README data

> wiring-pi@2.1.0 install /home/moma/node_modules/wiring-pi
> /bin/bash ./install.sh

Cloning libWiringPi ... failed.
--------------------------------------------
FATAL: Cloning libWiringPi failed.
Please check install.log and fix any problems. If you're still stuck,
then please open a new issue then post all the output and as many details as you can to
  https://github.com/eugeneware/wiring-pi/issues

-------------------------------------------
npm ERR! Linux 4.2.0-22-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "wiring-pi"
npm ERR! node v4.2.4
npm ERR! npm  v2.14.12
npm ERR! code ELIFECYCLE

npm ERR! wiring-pi@2.1.0 install: `/bin/bash ./install.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the wiring-pi@2.1.0 install script '/bin/bash ./install.sh'.
npm ERR! This is most likely a problem with the wiring-pi package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     /bin/bash ./install.sh
npm ERR! You can get their info via:
npm ERR!     npm owner ls wiring-pi
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/moma/npm-debug.log
moma@kursk1510:~$ 

BTW: I managed to Install Nodejs, npm and wiring-pi on Raspberry Pi2 with Rasbian distro.
Thanks in advance.

nekuz0r, 2016-01-05:
Can you attach the `install.log` file please, should be in `node_modules/wiring-pi`

osmoma, 2016-01-05:
My $HOME has no node_modules/ directory.  I checked this several times. users
$ find $HOME -name install.log
(not found)

However I found some wiring-pi directories. The first one is empty.
 $ find . -name wiring-pi
./.npm/registry.npmjs.org/wiring-pi
./.npm/wiring-pi
-----
$ tree ./.npm/registry.npmjs.org/wiring-pi
0 directories, 0 files
-----
$ tree ./.npm/wiring-pi
./.npm/wiring-pi
└── 2.1.0
    ├── package
    │   └── package.json
    └── package.tgz
-----
Thanx for your reply.  moma.

osmoma, 2016-01-05:
#SOLVED.
I downloaded the source code from Github and run the 
$ cd wiring-pi-master 
$ bash ./install.sh 

$ cat install.log
./install.sh: 43: ./install.sh: git: not found.   
LOL ;-)

$ sudo apt install git
$ sudo npm install wiring-pi 
OK!
[success1.zip](https://github.com/eugeneware/wiring-pi/files/78996/success1.zip)

nekuz0r, 2016-01-05:
Ok, great, nice you figured it out.

closed 2016-01-05 by nekuz0r