WiringPi/WiringPi-Node #38: Making gpio utility failed

chinajuanbob, 2016-01-22:
Got this problem while install it in raspberry pi 2

screen output:

root@raspberrypi:~/web# npm install -g wiring-pi
-
> wiring-pi@2.1.1 install /usr/lib/node_modules/wiring-pi
> /bin/bash ./install.sh

Cloning libWiringPi ... done.
patching file ./wiringpi/devLib/Makefile
Hunk #1 succeeded at 24 with fuzz 1 (offset -2 lines).
Hunk #2 succeeded at 32 with fuzz 2 (offset -2 lines).
patching file ./wiringpi/gpio/Makefile
Making libWiringPi ... done.
Making devLib ...done.
Unistalling gpio utility ... [sudo] password for nobody:
[sudo] password for nobody:
[sudo] password for nobody:
done.
Making gpio utility ... done.
Installing gpio utility ... [sudo] password for nobody:
[sudo] password for nobody:
[sudo] password for nobody:
failed.

================================================================================
Making gpio utility failed.
Please check install.log and fix any problems. If you're still stuck,
then please open a new issue then post all the output and as many details as you can to
  https://github.com/eugeneware/wiring-pi/issues
================================================================================

done.

How to solve it? What is the password for nobody?
Thanks!

chinajuanbob, 2016-01-22:
Issue "npm install wiring-pi" works fine, so global installation is not supported.

mefynn, 2016-10-31:
i get the same error. while in pi in root or pi with sudo...