WiringPi/WiringPi-Node #39: Add GPIO multiplexer support

nkolban, 2016-02-27:
I am coming across projects where I need to use more GPIO pins than are physically available on the Pi.  To solve my problems, I then use GPIO extenders such as PCF8574 and MCP23017.  Since these are addressable devices in their own right, I can't use the wiringPi APIs such as digitalWrite and digitalRead directly against them.

This proposal suggests the enhancement of the wiring-pi module to allow us to pass an object as the pin parameter for some of the APIs.  That object will be an interface (to be considered) that implements a GPIO extender set of parameters.  This will allow me to use our existing module but when I want to read or write from an extended pin, I can pass in an object that implements the accessor.  We will supply sample accessors for the common extenders.

I am happy to own or participate in this part of the project but want to pass it by the community for discussion and concurrence.   If the idea isn't desired then I'll obviously not push it any further.

Neil

DPD-30, 2016-03-05:
Those chips are already supported check the documentation.  Here is the [PCF8574 link](https://github.com/eugeneware/wiring-pi/blob/master/DOCUMENTATION.md#pcf8574)

nkolban, 2016-03-05:
Many thanks for the response.  Can you help clarify what we mean by the PCF8574 as an extension?

I am imagining having a PCF8574 .... and "somehow" wish to say that it represents some pins.

digitalWrite(myPCF8574.pin(0), true);

Ideally, I want to be able to pass an "object" or indicator as the first parameter of a digitalWrite that would "drive" a PCF8574 or other GPIO extender.

nekuz0r, 2016-03-05:
call `pcf8574Setup(pinBase, i2cAddress)` to initialize your GPIO extender
with the first parameter being your pin base for this chip to be mapped to and `i2cAddress` is the address of that chip on the I2C bus.

`pcf8574Setup(200, 0x30)`

then you will be able to access the first IO of the chip with the following call

`digitalWrite(200, HIGH)`

nkolban, 2016-03-05:
FANTASTIC!!!   This is exactly what I was looking for.   I'll study this function in detail and if I can assist with improving the documentation so others may find this themselves ... I'll certainly do that.  Again ... FANTASTIC ... exactly what I was hoping for.

closed 2016-03-05 by nkolban