WiringPi/WiringPi-Node #42: sn3218 extension not available

paulrosbif, 2016-04-28:
I'm using node v5.11.0 on a Raspberry PI 3, with gpio v2.32.  I've added the wiring-pi via npm and while I can use the basic functions (for example wpi.wiringPiSetup()) when I try to run a simple test:
`var wpi = require('wiring-pi');
var basePin = 64;
wpi.wiringPiSetup();
wpi.sn3218Setup(basePin);
wpi.analogWrite(1,255);
console.log("Pin should be on");`
I get `TypeError: wpi.sn3218Setup is not a function`.  Is there some additional setup I'm missing for the extension?

nekuz0r, 2016-09-21:
Hi, just checked this out, and looks like some extensions are not loaded in https://github.com/eugeneware/wiring-pi/blob/master/src/extensions/extensions.cc

NoHomey, 2016-09-29:
Hey @nekuz0r  I've just created a PR #49 to fix this issue. Could you review it?

closed 2016-09-29 by nekuz0r

paulrosbif, 2016-09-29:
> Closed #42 <https://github.com/eugeneware/wiring-pi/issues/42> via 6931958

Excellent! I'll go try it
Paul

commit 6931958c (2016-09-29): Merge pull request #49 from NoHomey/patch-3

Fixes #42 sn3218 extension not available