WiringPi/WiringPi-Node #44: Building on Windows

nVentis, 2016-08-05:
Is it possible to adjust the install script files in a way that allows building wiring-pi on Windows? Tried with Cygwin and / or MinGW on Windows 10, but it seems like the reference to /bin/bash is the main error here, which just can not be resolved by manipulating the PATH variable.

My aim is to use wiring-pi as part of a Node JS project aimed at Windows 10 IoT using NTVS (Node JS Tools for Visual Studio) together with an extension called NTVSIoT which basically represents an altered version of Node JS just with Microsoft's Chakra engine instead of V8.

dodekeract, 2016-08-06:
Did you try bash on windows?

nVentis, 2016-08-06:
Yes, and it fails with
```

User@Computer /cygdrive/d/DevRepositories/wiring-pi
$ /bin/bash install.sh
install.sh: line 2: $'\r': command not found
install.sh: line 3: syntax error near unexpected token `$'{\r''
'nstall.sh: line 3: `check_make_ok() {
```

nekuz0r, 2016-09-21:
I am not supporting windows, i don't even have a raspberry pi with windows 10 IoT installed.
But feel free to fix it and submit a pull request :)
In the mean time i am closing this issue.

closed 2016-09-21 by nekuz0r