WiringPi/WiringPi-Node #52: Release new sem. version

NoHomey, 2016-10-04:
I would like to request a new sem. version release. Since all commits after [2.1.1](https://github.com/eugeneware/wiring-pi/compare/v2.1.1...master) are result of mine pull requests beeing merged I would like to update both the README.md and CHANGELOG.md so new version could be released. But first #51 needs to be closed.

New version is needed because of the following issues have been fixed since `2.1.1`:

- Fixing in which `ints` `pinmodeAlt` `mode` is checked, [https://github.com/eugeneware/wiring-pi/commit/24b09083ac5842a914e3369d5af69a4a3bf4bcba]
- All `WiringPi` `WPI_MODE_*` constants are now exported, [https://github.com/eugeneware/wiring-pi/commit/5bea401d9738d995f5b65d3ec013f589f2ca1719]
- sn3218 extention has been made available, (closes #42), [https://github.com/eugeneware/wiring-pi/commit/6931958c392968dd53d06cc39b133a6d852d0f46]
- all file descriptors opened by `wiring-pi` are now releasable, (closes #43 ) [https://github.com/eugeneware/wiring-pi/commit/bf14d466b56761d5de40511549fbcd8b6c6ebe0c]

nekuz0r, 2016-10-05:
I will update the changelog accordingly to your fixes/additions this evening and bump the version 2.2.0

nekuz0r, 2016-10-05:
Thanks a lot for your contributions, much appreciated 💯

commit baea7afa (2016-10-06): Merge pull request #53 from NoHomey/2.2.0

Closes #52 Release new sem. version
closed 2016-10-06 by nekuz0r