WiringPi/WiringPi-Node #62: wiringPiSetup: Must be root. (Did you forget sudo?)

Jarvis785, 2016-10-27:
this is happening for what ?? i also unable to install to install using sudo npm install -g wiring-pi
beacuse its asking for password [sudo] password for nobody ??

Jarvis785, 2016-10-27:
here is the install log in /usr/local/lib/node_modules/wiring-pi

[install.log.txt](https://github.com/eugeneware/wiring-pi/files/555510/install.log.txt)

dodekeract, 2016-10-27:
Can you try `sudo -s` and `npm i -g wiring-pi`?

Jarvis785, 2016-10-27:
its installed now without errors but......
to run i have to write  $ sudo node something.js

its ok to type sudo in terminal and run ...

but when it comes to run at startup automatically .... then it becomes a problem....
its showing  :: wiringPiSetup: Must be root. (Did you forget sudo?)

can u say what password to type during global install (sudo npm install -g wiring-pi) ?? my pi's password or what ??

Jarvis785, 2016-10-27:
okey .. can u suggest the best .. robust ..and easy  gpio pin library for node.js ???  which one is best ??

dodekeract, 2016-10-27:
Per default, sudo has no password on a raspberry pi. At least that's what I experienced on mine.

Jarvis785, 2016-10-28:
how to run wiring-pi without sudo...
wiringPiSetup: Must be root. (Did you forget sudo?)   :(

nekuz0r, 2016-10-28:
1. Global installation is not supported (see #38)
2. It's your JS application that requires to be run as root (see #32)

closed 2016-10-28 by nekuz0r