WiringPi/WiringPi-Node #63: wiringPiISR as non root

paulish, 2016-11-02:
The documentation states that

wiringPiISR method can be called as non root
> This function will work in any mode, and does not need root privileges to work.

At the same time a call to this method fails until setup() which in its turn requires root privileges.
wiringPiSetup and friends have this in the documentation and my own tests shows that this is true.
>This function needs to be called with root privileges.

So how to be if running the node as root is not an option?

paulish, 2016-11-02:
Sorry, my fault: wiringPiSetupSys can be called as non-root.

Please close this issue.

TheNotary, 2016-11-09:
I was having some trouble getting it working in non-root mode as well.  Based on the [docs](https://github.com/eugeneware/wiring-pi/blob/master/DOCUMENTATION.md#apis).  I eventually realized I had overlooked ensuring that "the GPIO pins have been exported before-hand using gpio program."  

Here's code for non-root

```
/* 
  Pins:

  black wire  = header 6  = ground    
  orange wire = header 11 = wpi 0
  yellow      = header 13 = wpi 2
  red           = header 15 = wpi 3
*/

// Setup GPIO pins...
var exec = require('child_process').exec;
var cmd = 'gpio export 0 in && gpio export 2 in && gpio export 3 in';

exec(cmd, function(error, stdout, stderr) {
    // command output is in stdout
    //console.log(error);
    console.log(stdout);
    console.log(stderr);
});

// 1. Start listening for wiring_pi interupts

var wpi = require('wiring-pi');
//wpi.setup('wpi');
wpi.setup('sys');

var pin = 0;

wpi.pinMode(pin, wpi.INPUT);
wpi.pullUpDnControl(pin, wpi.PUD_UP)

var value = wpi.digitalRead(pin);

console.log("The pin state was: " + value);
```

closed 2016-11-21 by paulish