WiringPi/WiringPi-Node #68: npm install wiring-pi on jessie fails

siliconchris1973, 2017-02-17:
Hello All,

I have a Raspberry PI 2 with installed wiring-pi. Here is what gpio -v logs:

```
gpio -v
gpio version: 2.25
Copyright (c) 2012-2015 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Model 2, Revision: 1.1, Memory: 1024MB, Maker: Sony 

```
I want to use wiring-pi in a node.js program so I tried npm install wiring-pi and here it what happens:

```
npm install wiring-pi

> wiring-pi@2.2.1 install /data/shared/Developing/ledconodel/node_modules/wiring-pi
> /bin/bash ./install.sh

Cloning libWiringPi ... done.
patching file ./wiringpi/devLib/Makefile
Hunk #1 succeeded at 24 with fuzz 1 (offset -2 lines).
Hunk #2 succeeded at 32 with fuzz 2 (offset -2 lines).
patching file ./wiringpi/gpio/Makefile
Making libWiringPi ... failed.

================================================================================
FATAL: Making libWiringPi failed.
Please check install.log and fix any problems. If you're still stuck,
then please open a new issue then post all the output and as many details as you can to
  https://github.com/WiringPi/WiringPi-Node/issues
================================================================================

npm ERR! Linux 4.4.38-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "wiring-pi"
npm ERR! node v7.5.0
npm ERR! npm  v4.1.2
npm ERR! code ELIFECYCLE

npm ERR! wiring-pi@2.2.1 install: `/bin/bash ./install.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the wiring-pi@2.2.1 install script '/bin/bash ./install.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the wiring-pi package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     /bin/bash ./install.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs wiring-pi
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls wiring-pi
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /data/shared/Developing/ledconodel/npm-debug.log

```

I also attached the full npm debug log in hope someone can tell me whats going wrong.

Please, could someone help me?

Best regards,

Chris

[npm-debug.log.zip](https://github.com/WiringPi/WiringPi-Node/files/782945/npm-debug.log.zip)

siliconchris1973, 2017-02-17:
I have additional information on this issue.
I tried to do a manual installation of wiring-pi from the base package wiring-pi-2.2.1 as was fetch from npm registry. 

During running install.sh I noticed a segmentation fault as can be seen in below snippet:

`gcc: internal compiler error: **Segmentation fault** (program as)`

Funny to say, I already had wiring-pi running on this PI the last couple of days. Then I removed all node_modules and started from scratch.

VERY STRANGE!

Here is the output of the install.sh script:

```
bash install.sh 
Cloning libWiringPi ... done.
patching file ./wiringpi/devLib/Makefile
Hunk #1 succeeded at 24 with fuzz 1 (offset -2 lines).
Hunk #2 succeeded at 32 with fuzz 2 (offset -2 lines).
patching file ./wiringpi/gpio/Makefile
Making libWiringPi ... failed.

================================================================================
FATAL: Making libWiringPi failed.
Please check install.log and fix any problems. If you're still stuck,
then please open a new issue then post all the output and as many details as you can to
  https://github.com/WiringPi/WiringPi-Node/issues
================================================================================
```

Here you'll find the output of install.log:

```
cat install.log 
Cloning into 'wiringpi'...
Note: checking out '6d98e0c87ad937849af944ae7118ce1e01ce7c91'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

[Clean]
[Compile] wiringPi.c
gcc: internal compiler error: Segmentation fault (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
Makefile:92: recipe for target 'wiringPi.o' failed
make: *** [wiringPi.o] Error 4

```

siliconchris1973, 2017-02-17:
For some reason I cannot grasp at the moment, an 

```
apt-get update
apt-get upgrade
apt-get remove nodejs
apt-get purge
apt-get install nodejs
```
and then a reboot plus again removing all node modules, fixed this issue.

I'm closing it, but if anyone can find the original reason, I'd really appreciate a message.

BR,

Chris

closed 2017-02-17 by siliconchris1973

viktorbk, 2019-03-21:
The problem here is that V8 has removed ForceSet method in the V8 version that Node 10 uses.
Solution is to install older version of nodejs. Easiest is to use NodeVersionManager nvm.
curl -o- https://raw.githubusercontent.com/creationix/nvm/v0.33.8/install.sh | bash
source ~/.bashrc 
nvm install 8
npm install npm@latest -g

then you can do 

npm remove wiring-pi
npm install wiring-pi