WiringPi/WiringPi-Node #70: New version to support Raspbian kernel 4.9.y ?

gundrygs, 2017-03-06:
Hi,
I see that the guys @ https://github.com/raspberrypi/linux have changed the Linux headers default branch to 4.9.y, and are expecting this to be the new Long Term Support (LTS) version. This causes an issue for wiring-pi when using a 4.9.y Raspbian kernel, as the kernel (correctly) reports the hardware on a Pi3 as 'BCM2835', but this generates an error message regarding unsupported hardware in wiring-pi. It appears Gordon Henderson fixed this issue some time ago in the C library for wiringPi (http://wiringpi.com), but your Node wrapper builds against a rather out of date mirror at https://github.com/nekuz0r/wiringpi.git.
Do you plan to resolve this ?
Regards,
George

amsross, 2017-04-29:
I took a stab at changing the target to a more up-to-date WiringPi lib, but I'm seeing failure at `node-gyp rebuild`. I don't have much experience making/updating binding libraries, so I'm not sure where to go from there.

https://github.com/amsross/WiringPi-Node

gundrygs, 2017-04-29:
Hi Ross,

I have had success replacing the "wiring-pi" library with the "wiringpi-node" library - just had to npm install and change the require statement in my code - worked on the new kernel without further modifications.

See https://www.npmjs.com/package/wiringpi-node

Hope this helps,

George

kohei0302, 2017-05-22:
I forked this repository to use the latest library.

https://github.com/kohei0302/WiringPi-Node
https://github.com/kohei0302/wiringpi

RandScullard, 2017-05-29:
Is anyone working on resolving this issue for real? I was able to use @kohei0302's fork to get running again (thanks for that!), but that's no substitute for a real fix. I can't tell if anyone is actively maintaining this project  -- @eugeneware are you on top of this? Has this project been superseded by a different node.js wrapper for WiringPi?

val34m, 2017-08-06:
Hello

I am trying to install wiring-pi module for nodejs but I am facing some problms.
Here you can see all I have tried:
-Install  Raspbian laste version Version:July 2017 Release date:2017-07-05 Kernel version:4.9
-Make an sudo apt-get update / upgrade
-sudo apt-get install npm.
-npm install wiring-pi
-npm install wiringpi-node
-node - v
v0.10.29
-npm - v
1.4.21

when I run a in /home/pi/node-modules/wiringpi-node/examples:
sudo nodejs bink.js
I have the same problem with the BCM2835.

I had install wiringPi lib with the last version.  I have no problem in command line with gpio cmd, but that doesn't work with node.
In the package wiring-pi and wiringpi-node the version of wiringpi is 2.44

What am I doing wrong ? I don't understand.
I read the issue problem with the "Bump to wiringpi-node 2.4.4" but i dont understand what to do and where are files to modify.

Thank a lot for your help.

gundrygs, 2017-08-06:
I only npm installed 'wiringpi-node', not both libraries - note that you will need to change the 'require' statement in your code - for example, change "const wPi = require('wiring-pi')" to "const wPi = require('wiringpi-node')"

Without this change in both your code and the blink example, you will continue to reference the original wiring-pi library, and so continue to see the problem.

Hope this helps.

val34m, 2017-08-06:
Hello thank for your answer.

I had reinstall all the raspbian next npm and wiringpi-node with npm.
In the file /home/pi/node_modules/wiringpi-node/examples/blink.js
I change the require : 
var wpi = require('wiring-pi');
By:
var wpi = require('wiringpi-node');

I have this error : 
pi@raspberrypi:~/node_modules/wiringpi-node/examples $ sudo node blink.js

module.js:356
  Module._extensions[extension](this, filename);
                               ^
Error: Symbol WiringPi_module not found.
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/pi/node_modules/wiringpi-node/lib/exports.js:1:80)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)

Thank for your help

gundrygs, 2017-08-06:
Have you tried copying blink.js to your /home/pi directory and running 'sudo node blink.js' from there ?

val34m, 2017-08-06:
Hello

I tried to create a new file name test.js

I just have written the require statment And i have the same issue

Is node version v0.10.29 is good ?

Thank

val34m, 2017-08-07:
Hello

After many tests, i have found what didn't worked.
we need to use node version 6 and not v0.10.29.

blink.js work fine.

Thank for the support, and for the great job

gundrygs, 2017-08-07:
I am very pleased you managed to "crack the case" 😊

val34m, 2017-08-08:
Hello 

I 'have an other problem :(

I tried to configure a pca9685 module so this is what i tried : 
=> create a file pca.js
In this file : 
var wpi=require('wiringpi-node');
wpi.pca9685Setup(100,0x40,50);

When i tried this i had : 
TypeError: wpi.pca9685Setup is not a function

SO i tried to liisted all fonctions in wpi with a 
console.log(wpi);
I can see they have no fonction pca9685 

How to use pca9685 module ?
I have se same problem for mcp23017

Thank

RandScullard, 2017-08-08:
WiringPi doesn't have drivers for specific devices. You need to either find a node.js driver for your device or write your own. It looks like there are existing drivers for the pca9685 (https://github.com/johntreacy/adafruit-pca9685) and mcp23017 (https://github.com/kaihenzler/node-mcp23017) so you should check those out. I haven't used either of these devices so I can't vouch for the drivers. For the most part I've found that I've had to write my own drivers for the devices I've used so far; often this is a matter of porting existing Python code to JavaScript.

Interesting note: Neither of the drivers mentioned above actually uses WiringPi-Node.

val34m, 2017-08-09:
Hello

Thank for the answer.
I had asked that beacause in the documentation they have mentionned in extension part these modules

kishu27, 2017-09-07:
bumping this issue. Maintainers, please consider this issue as it's being a blocker now

kishu27, 2017-09-07:
@kohei0302 could you please comment how can I use your repository instead of this one to get around this issue? Thanks!

NoHomey, 2017-09-08:
I'm going to rewrite the whole project. I had alot of ideas to make the project more open for me and other contributors by moving it to an open organization see #61. I also opened an issue #60 about moving to NAN. But just as I started my work on rewriting last year I received a life changing opportunity for me which I could not reject. On top of that I'm currently studying mathematics and computer science verry seriously, so I had no time for this project.  Now I have whole month free just for me and I'll use it to contribute to project like this one ;) Sorry for he novel about my life but I have to explain why #60 is still open ...

kishu27, 2017-09-08:
Haha.. totally understand :) 
If for the time being, it's not so much of a problem, maybe you can consider merging https://github.com/kohei0302/WiringPi-Node

NoHomey, 2017-09-08:
@RandScullard I've checked both repos. They are using only I2C bus. [i2c](https://github.com/kelly/node-i2c) npm package they use ships with [I2C device driver](https://github.com/kelly/node-i2c/tree/master/src) this way it has an interface talking directly to the hardware and so no need to use WiringPi lib ...

NoHomey, 2017-09-08:
@kishu27 I'll write to @kohei0302 asking him to make a PR which I can than merge.

NoHomey, 2017-09-08:
@kishu27 I have not found any contact to @kohei0302 so I made the PR with which to merge the commits from his fork. The problem is that @kohei0302 's fork builds against his [fork of wiringpi](https://github.com/kohei0302/wiringpi) as well ... So I'wont merge the PR before @nekuz0r  reviews it.