WiringPi/WiringPi-Node #72: Does this support orange pi zero?

justin-romano, 2017-03-21:
Im getting this
```
root@orangepizero:~/src/node-wiring-op/examples# nodejs pwm.js
Unable to determine hardware version. I see: Hardware   : sun8i
,
 - expecting BCM2708 or BCM2709. Please report this to projects@drogon.net
root@orangepizero:~/src/node-wiring-op/examples#

```

dodekeract, 2017-03-21:
While it may be possible to add support, this message obviously means that it's currently not supported. :/

justin-romano, 2017-03-21:
But is must be close, not sure how different the H2 and H3 are but maybe it would just work if we added the new hardware to the probe

shoutrain, 2017-05-19:
@justin-romano I am using H2, OrangePi Zero, it should works with WiringPI, but different num.  Following the map output by WiringOP-Zero
```
 +-----+-----+----------+------+--Orange Pi Zero--+---+------+---------+-----+--+
 | H2+ | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | H2+ |
 +-----+-----+----------+------+---+----++----+---+------+----------+-----+-----+
 |     |     |     3.3v |      |   |  1 || 2  |   |      | 5v       |     |     |
 |  12 |   8 |    SDA.0 | ALT5 | 0 |  3 || 4  |   |      | 5V       |     |     |
 |  11 |   9 |    SCL.0 | ALT5 | 0 |  5 || 6  |   |      | 0v       |     |     |
 |   6 |   7 |   GPIO.7 | ALT3 | 0 |  7 || 8  | 0 | ALT5 | TxD3     | 15  | 198 |
 |     |     |       0v |      |   |  9 || 10 | 0 | ALT5 | RxD3     | 16  | 199 |
 |   1 |   0 |     RxD2 | ALT5 | 0 | 11 || 12 | 0 | ALT3 | GPIO.1   | 1   | 7   |
 |   0 |   2 |     TxD2 | ALT5 | 0 | 13 || 14 |   |      | 0v       |     |     |
 |   3 |   3 |     CTS2 | ALT3 | 0 | 15 || 16 | 0 | ALT4 | GPIO.4   | 4   | 19  |
 |     |     |     3.3v |      |   | 17 || 18 | 0 | ALT4 | GPIO.5   | 5   | 18  |
 |  15 |  12 |     MOSI | ALT3 | 0 | 19 || 20 |   |      | 0v       |     |     |
 |  16 |  13 |     MISO | ALT3 | 0 | 21 || 22 | 0 | ALT3 | RTS2     | 6   | 2   |
 |  14 |  14 |     SCLK | ALT3 | 0 | 23 || 24 | 0 | ALT3 | CE0      | 10  | 13  |
 |     |     |       0v |      |   | 25 || 26 | 1 | OUT  | GPIO.11  | 11  | 10  |
 +-----+-----+----------+------+---+---LEDs---+---+------+----------+-----+-----+
 |  17 |  30 | STAT-LED |  OUT | 1 | 27 || 28 |   |      | PWR-LED  |     |     |
 +-----+-----+----------+------+---+-----+----+---+------+----------+-----+-----+
 | H2+ | wPi |   Name   | Mode | V | Physical | V | Mode | Name     | wPi | H2+ |
 +-----+-----+----------+------+--Orange Pi Zero--+---+------+---------+-----+--+
```

paduvi, 2017-06-17:
I'm using NanoPI M1. It displays the same message. I think they're using the same core library, just different header num.