WiringPi/WiringPi-Node #73: Does not compile on Node v7.7.3

pensierinmusica, 2017-03-24:
Trying to install on Node v7.7.3, I get:

```sh
FATAL: Making libWiringPi failed.
```

And in `npm-debug.log`:

```sh
441 error Failed at the wiring-pi@2.2.1 install script '/bin/bash ./install.sh'.
```

Any idea how to fix this? Thx!

pensierinmusica, 2017-03-24:
Tried on Node v6.9.2 and I get the same problem.

vabovyan, 2017-04-05:
sudo apt-get install git-core 
will fix it.

pensierinmusica, 2017-04-10:
`apt-get` does not exist on Mac, I think it's for Linux. Maybe with homebrew?

caiusCitiriga, 2018-01-25:
same problem here.

viktorbk, 2019-03-21:
The problem here is that V8 has removed ForceSet method in the V8 version that Node 10 uses.
Solution is to install older version of nodejs. Easiest is to use NodeVersionManager nvm.
curl -o- https://raw.githubusercontent.com/creationix/nvm/v0.33.8/install.sh | bash
source ~/.bashrc 
nvm install 8
npm install npm@latest -g

then you can do 

npm remove wiring-pi
npm install wiring-pi