WiringPi/WiringPi-Node #76: how to build for firmware 4.9 and wiringpi 2.4.4?

ylh888, 2017-04-30:
When I built wiringpi from scratch, I get version 2.4.4. gpio -v yields:
``` 
gpio version: 2.44
Copyright (c) 2012-2017 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 2, Revision: 01, Memory: 1024MB, Maker: Embest
  * Device tree is enabled.
  *--> Raspberry Pi 2 Model B Rev 1.1
  * This Raspberry Pi supports user-level GPIO access.
```

When I use "npm install" to get the node bindings, the install script rebuilds wiringpi using 2.25 
```
gpio version: 2.25
Copyright (c) 2012-2015 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Unable to determine hardware version. I see: Hardware	: BCM2835
,
 - expecting BCM2708 or BCM2709.
```
This does not seem to be compatible with the firmware because "sudo node blink.js" yields:
```
sudo node blink.js
Unable to determine hardware version. I see: Hardware	: BCM2835
,
 - expecting BCM2708 or BCM2709.
```

Is it possible to use the latest wiringpi libraries with the node bindings?

pradhanshrijal, 2017-05-11:
Having the same problem...will this problem get solved or not?

ylh888, 2017-05-11:
take a look at a solution from this issue: https://github.com/WiringPi/WiringPi-Node/issues/70

xmxstudio, 2017-07-04:
thanks for the link to #70 however that's not a valid "fix" 

I'm also having this issue. Was working fine until dist-upgrade and everything went to shit

eczajk1, 2017-11-14:
I'm having similar issue when trying to run `wiring-pi` in a [Resin](https://resin.io/) container:

```sh
13.11.17 13:15:16 (-0500) Updating application 'registry2.resin.io/redactedapp/aad287ad7be2b5e094f047edd41fdcc788632b79'
13.11.17 13:15:16 (-0500) Killing application 'registry2.resin.io/redactedapp/aad287ad7be2b5e094f047edd41fdcc788632b79'
13.11.17 13:15:16 (-0500) Sending SIGTERM to remaining processes...
13.11.17 13:15:16 (-0500) Sending SIGKILL to remaining processes...
13.11.17 13:15:16 (-0500) Unmounting file systems.
13.11.17 13:15:16 (-0500) Unmounting /sys/kernel/debug.
13.11.17 13:15:16 (-0500) Unmounting /dev/mqueue.
13.11.17 13:15:16 (-0500) All filesystems unmounted.
13.11.17 13:15:16 (-0500) Halting system.
13.11.17 13:15:17 (-0500) Application exited 'registry2.resin.io/redactedapp/aad287ad7be2b5e094f047edd41fdcc788632b79'
13.11.17 13:15:18 (-0500) Killed application 'registry2.resin.io/redactedapp/aad287ad7be2b5e094f047edd41fdcc788632b79'
13.11.17 13:15:18 (-0500) Installing application 'registry2.resin.io/redactedapp/aad287ad7be2b5e094f047edd41fdcc788632b79'
13.11.17 13:15:20 (-0500) Installed application 'registry2.resin.io/redactedapp/aad287ad7be2b5e094f047edd41fdcc788632b79'
13.11.17 13:15:20 (-0500) Starting application 'registry2.resin.io/redactedapp/aad287ad7be2b5e094f047edd41fdcc788632b79'
13.11.17 13:15:21 (-0500) Started application 'registry2.resin.io/redactedapp/aad287ad7be2b5e094f047edd41fdcc788632b79'
13.11.17 13:15:22 (-0500) Systemd init system enabled.
13.11.17 13:15:22 (-0500) systemd 215 running in system mode. (+PAM +AUDIT +SELINUX +IMA +SYSVINIT +LIBCRYPTSETUP +GCRYPT +ACL +XZ -SECCOMP -APPARMOR)
13.11.17 13:15:22 (-0500) Detected virtualization 'other'.
13.11.17 13:15:22 (-0500) Detected architecture 'arm'.
13.11.17 13:15:22 (-0500) Set hostname to <7c539c2>.

13.11.17 13:15:31 (-0500) > redacted@0.0.0 start /usr/src/app
13.11.17 13:15:31 (-0500) > node ./app/app.js
13.11.17 13:15:31 (-0500)
13.11.17 13:15:34 (-0500) Unable to determine hardware version. I see: Hardware : BCM2835
13.11.17 13:15:34 (-0500) ,
13.11.17 13:15:34 (-0500)  - expecting BCM2708 or BCM2709. Please report this to projects@drogon.net
13.11.17 13:15:34 (-0500)
```

spielhoelle, 2018-04-28:
As @ylh888 recommended in #70 - following solution worked for me:

> Hi Ross,
> I have had success replacing the "wiring-pi" library with the "wiringpi-node" library - just had to npm install and change the require statement in my code - worked on the new kernel without further modifications.
> See https://www.npmjs.com/package/wiringpi-node
>
> Hope this helps,
>
> George