WiringPi/WiringPi-Node #77: example for pwmWrite?

hnoesekabel, 2017-05-14:
Hi,

apologies for a newcomer question, but I was unable to find example code how to pwmWrite to a LED. My setup looks correct, and running Sunfounder Python code (04_pwmLed.py from the Superkit 2 code snippets) lets the LED pulse as expected.

With WiringPi-Node, I'm running the following snippet on my Pi 3:

```
'use strict';
var wpi = require('wiringpi-node');
wpi.setup('wpi');

wpi.pinMode(1, wpi.PWM_OUTPUT);
wpi.pwmWrite(1, 0);
```

...which lights up the LED but as soon as I run it, my Rasperry Ext4 filesystem crashes and the only option is to pull the plug. Google results suggest that these Ext4 FS errors occur due to power supply issues - as they do not occur with the Python code, and it's an power supply coming with a standard Rasperry Pi kit, I'd assume it's not hardware related.

Any ideas what's wrong? Any initialisation I skipped? Any code samples I could try? Is there a repository of more WiringPi-Node examples? Any pointers are appreciated.

Thanks!

carterw, 2017-05-15:
Do you have a resistor in series with the LED?

RichardChambers, 2017-11-26:
Are you running it with sudo? I ran into a similar problem with C using the PWM with GWPIO #18 in that I had a program using PWM with an LED circuit and my Raspberry Pi with Raspbian crashed every time I tried to run it.

Then I discovered I needed to use sudo to run the program because the wiringPi library required administrator or root access to the PWM hardware.

michaeljanich, 2017-12-06:
Sounds like a power problem to me. Get a bigger power source (1.5A) 

And yes, have a 47Ohm resistor in series with the LED.