WiringPi/WiringPi-Node #88: Unable to determine hardware version: I see: hardware: bcm2835

andy977, 2018-01-25:
Please update the install script to download the latest version of wiringpi from their git repo as the version you currently have  in the script does not work on linux kernel 4.9.x. There are other wrappers on top of wiringPi that have the same issue.

tillepille, 2018-01-25:
Just as an idea to this problem: As how I see it, why not pull directly from git.drogon.net/wiringPi than use these unregular maintained repos?

daintycode, 2018-03-13:
Documentation is aware of bcm2835, but the software isn't?
[DOCUMENTATION.md:195](https://github.com/WiringPi/WiringPi-Node/blame/master/DOCUMENTATION.md#L195) : "Unlike Arduino, the **BCM2835** has both pull-up and down internal resistors."

gouthamgandhi, 2018-03-18:
I see the same problem anyone found a work around for this ?

toolboc, 2018-04-20:
You may try downgrading your raspberry pi kernel to 4.4.50, this fixed the issue for me.  

`sudo rpi-update 52241088c1da59a359110d39c1875cda56496764`

iammoen, 2019-04-03:
> You may try downgrading your raspberry pi kernel to 4.4.50, this fixed the issue for me.
> 
> `sudo rpi-update 52241088c1da59a359110d39c1875cda56496764`

This worked for me. One note is that I had to remove the node_modules folder and do an `npm install` after the reboot otherwise I would get a Segmentation Fault. Just in case someone else stumbles across that.