WiringPi/WiringPi-Node #89: can't install wiringpi on raspy3

Trigun87, 2018-02-15:
when i try to install wiringpi it always fail

`npm install wiring-pi

> wiring-pi@2.2.1 install /home/pi/node_modules/wiring-pi
> /bin/bash ./install.sh

Cloning libWiringPi ... done.
patching file ./wiringpi/devLib/Makefile
Hunk #1 succeeded at 24 with fuzz 1 (offset -2 lines).
Hunk #2 succeeded at 32 with fuzz 2 (offset -2 lines).
patching file ./wiringpi/gpio/Makefile
Making libWiringPi ... done.
Making devLib ...done.
Unistalling gpio utility ... done.
Making gpio utility ... done.
Installing gpio utility ... done.
Making wiring-pi ... make: Entering directory '/home/pi/node_modules/wiring-pi/build'
  CXX(target) Release/obj.target/wiringPi/src/addon.o
  CXX(target) Release/obj.target/wiringPi/src/wiringPi.o
In file included from ../src/wiringPi.h:4:0,
                 from ../src/wiringPi.cc:1:
../src/wiringPi.cc: In function ‘void nodemodule::initwiringPi(v8::Isolate*, v8::Handle<v8::Object>)’:
../src/addon.h:46:105: error: no matching function for call to ‘v8::Object::ForceSet(v8::Local<v8::String>, v8::Local<v8::Integer>, v8::PropertyAttribute)’
         v8::Int32::New(isolate, name), static_cast<v8::PropertyAttribute>(v8::ReadOnly | v8::DontDelete));
                                                                                                         ^
../src/wiringPi.cc:714:3: note: in expansion of macro ‘EXPORT_CONSTANT_INT’
   EXPORT_CONSTANT_INT(WPI_MODE_PINS);
   ^~~~~~~~~~~~~~~~~~~
In file included from /home/pi/.node-gyp/9.5.0/include/node/v8.h:26:0,
                 from ../src/addon.h:4,
                 from ../src/wiringPi.h:4,
                 from ../src/wiringPi.cc:1:
/home/pi/.node-gyp/9.5.0/include/node/v8.h:3115:29: note: candidate: v8::Maybe<bool> v8::Object::ForceSet(v8::Local<v8::Context>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)
                 Maybe<bool> ForceSet(Local<Context> context, Local<Value> key,
                             ^
/home/pi/.node-gyp/9.5.0/include/node/v8config.h:318:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
/home/pi/.node-gyp/9.5.0/include/node/v8.h:3115:29: note:   no known conversion for argument 3 from ‘v8::PropertyAttribute’ to ‘v8::Local<v8::Value>’
                 Maybe<bool> ForceSet(Local<Context> context, Local<Value> key,
                             ^
/home/pi/.node-gyp/9.5.0/include/node/v8config.h:318:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
In file included from ../src/wiringPi.h:4:0,
                 from ../src/wiringPi.cc:1:
../src/addon.h:46:105: error: no matching function for call to ‘v8::Object::ForceSet(v8::Local<v8::String>, v8::Local<v8::Integer>, v8::PropertyAttribute)’
         v8::Int32::New(isolate, name), static_cast<v8::PropertyAttribute>(v8::ReadOnly | v8::DontDelete));
                                                                                                         ^
../src/wiringPi.cc:715:3: note: in expansion of macro ‘EXPORT_CONSTANT_INT’
   EXPORT_CONSTANT_INT(WPI_MODE_PHYS);
   ^~~~~~~~~~~~~~~~~~~
In file included from /home/pi/.node-gyp/9.5.0/include/node/v8.h:26:0,
                 from ../src/addon.h:4,
                 from ../src/wiringPi.h:4,
                 from ../src/wiringPi.cc:1:
/home/pi/.node-gyp/9.5.0/include/node/v8.h:3115:29: note: candidate: v8::Maybe<bool> v8::Object::ForceSet(v8::Local<v8::Context>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)
                 Maybe<bool> ForceSet(Local<Context> context, Local<Value> key,
                             ^
/home/pi/.node-gyp/9.5.0/include/node/v8config.h:318:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
/home/pi/.node-gyp/9.5.0/include/node/v8.h:3115:29: note:   no known conversion for argument 3 from ‘v8::PropertyAttribute’ to ‘v8::Local<v8::Value>’
                 Maybe<bool> ForceSet(Local<Context> context, Local<Value> key,
                             ^
/home/pi/.node-gyp/9.5.0/include/node/v8config.h:318:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
In file included from ../src/wiringPi.h:4:0,
[... 1100 lines cut ...]
npm WARN pi No license field.

+ wiring-pi@2.2.1
updated 1 package in 29.021s
`
and  when i try to launch it 
` sudo node remote.js
module.js:557
    throw err;
    ^

Error: Cannot find module '../build/Release/wiringPi'
    at Function.Module._resolveFilename (module.js:555:15)
    at Function.Module._load (module.js:482:25)
    at Module.require (module.js:604:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/pi/node_modules/wiring-pi/lib/exports.js:1:80)
    at Module._compile (module.js:660:30)
    at Object.Module._extensions..js (module.js:671:10)
    at Module.load (module.js:573:32)
    at tryModuleLoad (module.js:513:12)
    at Function.Module._load (module.js:505:3)
`

how i can solve it?

ManfredSteiner, 2018-02-23:
I had similar problems on my Raspberry Pi 2 Model B v1.1. After switching back to kernel 4.4 (using command `rpi-update 52241088c1da59a359110d39c1875cda56496764`) and switching to Node.js v8.9.4 it could be installed and it was working. 

Installing [wiringpi-node](https://www.npmjs.com/package/wiringpi-node) instead of [wiring-pi](https://www.npmjs.com/package/wiring-pi) was also working on kernel 4.14.20-v7+ (using command `rpi-update` to update kernel to latest version).

That's not a solution, but maybe it helps...

AckerApple, 2018-02-27:
Thank you @ManfredSteiner

Could not get pi3 to run using wiring-pi. Switch my require to wiringpi-node and all is working now.

Awesome