WiringPi/WiringPi-Node #95: Can't install wiring pi

CopZabov, 2018-04-29:
I tried to install wiring-pi, i used every kind of procedure, but the result is always this. Can someone help me?
npm install wiring-pi

> wiring-pi@2.2.1 install /home/pi/.node-red/node_modules/wiring-pi
> /bin/bash ./install.sh

Cloning libWiringPi ... failed.

================================================================================
FATAL: Cloning libWiringPi failed.
Please check install.log and fix any problems. If you're still stuck,
then please open a new issue then post all the output and as many details as you can to
  https://github.com/WiringPi/WiringPi-Node/issues
================================================================================

npm WARN node-red-project@0.0.1 No repository field.
npm WARN node-red-project@0.0.1 No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! wiring-pi@2.2.1 install: `/bin/bash ./install.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the wiring-pi@2.2.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-04-29T15_47_38_327Z-debug.log

robeeeert, 2018-09-30:
Do you have `git` installed on your RPi?

michaeljanich, 2018-11-02:
Does anyone know of an excellent new wiringpi node package, that supports node10+?

jhoughjr, 2019-01-17:
dunno. I've spent all afternoon reading tons of useless docs, only to end up here. Seems wiring pi doesn't care about node since all the repos are 4 years old. Let me know if you find something that works. I'm just going to call my c program from node I guess.

Uweklaus, 2019-01-23:
same error msg here, while installing on my pizero with the newest node-red and node Versions.

Uweklaus, 2019-01-23:
installing according to wiring-pi page description works fine.
They are saying do not install wiring pi from here.

jhoughjr, 2019-01-24:
I may have borked my install. I switched to another pi and just ran install and rebuild and it worked fine. One difference is I used the rasping stretch full image in the latter.

viktorbk, 2019-03-21:
The problem here is that V8 has removed ForceSet method in the V8 version that Node 10 uses.
Solution is to install older version of nodejs. Easiest is to use NodeVersionManager nvm.
curl -o- https://raw.githubusercontent.com/creationix/nvm/v0.33.8/install.sh | bash
source ~/.bashrc 
nvm install 8
npm install npm@latest -g

then you can do 

npm remove wiring-pi
npm install wiring-pi

Fahad-pnw, 2019-04-22:
@viktorbk still won't work at the latest 8.series node..
anyway, now switched to 8.2.0 and testing it out, hopeless though

Fahad-pnw, 2019-04-22:
okay seems to have worked with 8.2.0
with a lots of warnings..but i guess it has installed successfully and would probably work.

../src/wiringPi.cc:773:3: note: in expansion of macro ‘EXPORT_CONSTANT_INT’
   EXPORT_CONSTANT_INT(FSEL_ALT2);
   ^~~~~~~~~~~~~~~~~~~
../src/addon.h:45:15: error: ‘class v8::Object’ has no member named ‘ForceSet’
       target->ForceSet(v8::String::NewFromUtf8(isolate, #name, v8::String::kInternalizedString), \
               ^
../src/wiringPi.cc:774:3: note: in expansion of macro ‘EXPORT_CONSTANT_INT’
   EXPORT_CONSTANT_INT(FSEL_ALT3);
   ^~~~~~~~~~~~~~~~~~~
../src/addon.h:45:15: error: ‘class v8::Object’ has no member named ‘ForceSet’
       target->ForceSet(v8::String::NewFromUtf8(isolate, #name, v8::String::kInternalizedString), \
               ^

tzwsoho, 2019-04-28:
> The problem here is that V8 has removed ForceSet method in the V8 version that Node 10 uses.
> Solution is to install older version of nodejs. Easiest is to use NodeVersionManager nvm.
> curl -o- [...]

I solved this problem with
git clone https://github.com/WiringPi/WiringPi-Node
cd WiringPi-Node
chmod +x install.sh
./install.sh

my OS Raspbian GNU/Linux 9 (stretch), with Raspberry 3B+

Pecacheu, 2019-06-15:
Okay, I looked though the v8 docs on how to fix this. Downgrading is not an option for me as the other packages I'm using require node 9 or even node 10 or higher.

I'm going to try to create a pull request replacing all instances of `ForceSet` with `DefineOwnProperty`. As far as I can tell the only difference is it requires a v8 context as the first argument.

Pecacheu, 2019-06-15:
Well, I tried, and solved the `ForceSet` errors, however I still get many mystery errors I can't solve. I cut out all the cascading warnings and errors caused by these three. The first I'm completely confused by. `target` doesn't appear to be in *wpi.cc* at first, but upon following the macro to *addon.h*, `target` is declared right here as a function parameter upon expansion:
```c
#define NODE_MODULE_INIT() \
namespace nodemodule { \
  void init(v8::Handle<v8::Object> target); \
} \
void nodemodule::init(v8::Handle<v8::Object> target)
// { Other Stuff }
#define INIT(name) nodemodule::init##name(isolate, target);
```
And `NODE_MODULE_INIT` gets expanded here in *wpi.cc:*
```c
NODE_MODULE_INIT() {
  #if NODE_VERSION_AT_LEAST(0, 11, 0)
    v8::Isolate* isolate = v8::Isolate::GetCurrent();
  #endif
  INIT(wiringPi);
  INIT(softPwm);
  INIT(softServo);
  INIT(softTone);
  INIT(wiringPiI2C);
  INIT(wiringPiSPI);
  INIT(wiringSerial);
  INIT(wiringShift);
  INIT(wiringPiISR);
  
  INIT(extensions);
  INIT(devlib);
}

NODE_MODULE_DECLARE(wiringPi);
```
Yeah, confusing, I know. This is why I hate excessive use of compiler macros. But regardless, that C code should work fine, so no clue where the error's coming from. Here are the errors:
```
../src/wpi.cc: In function ‘void node_register_module_v64(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Context>)’:
../src/addon.h:84:56: error: ‘target’ was not declared in this scope
     #define INIT(name) nodemodule::init##name(isolate, target);
                                                        ^
../src/wpi.cc:11:3: note: in expansion of macro ‘INIT’
   INIT(wiringPi);
   ^

In file included from ../src/wpi.cc:5:0:
../src/wpi.cc: At global scope:
/home/pi/.cache/node-gyp/10.16.0/include/node/node.h:567:30: error: redefinition of ‘node::node_module _module’
     static node::node_module _module =                                \
                              ^
/home/pi/.cache/node-gyp/10.16.0/include/node/node.h:607:3: note: in expansion of macro ‘NODE_MODULE_X’
   NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
   ^
../src/addon.h:77:39: note: in expansion of macro ‘NODE_MODULE’
     #define NODE_MODULE_DECLARE(name) NODE_MODULE(name, nodemodule::init)
                                       ^
../src/wpi.cc:25:1: note: in expansion of macro ‘NODE_MODULE_DECLARE’
 NODE_MODULE_DECLARE(wiringPi);
 ^
/home/pi/.cache/node-gyp/10.16.0/include/node/node.h:586:30: note: ‘node::node_module _module’ previously defined here
     static node::node_module _module =                                \
                              ^
/home/pi/.cache/node-gyp/10.16.0/include/node/node.h:611:3: note: in expansion of macro ‘NODE_MODULE_CONTEXT_AWARE_X’
   NODE_MODULE_CONTEXT_AWARE_X(modname, regfunc, NULL, 0)
   ^
/home/pi/.cache/node-gyp/10.16.0/include/node/node.h:634:3: note: in expansion of macro ‘NODE_MODULE_CONTEXT_AWARE’
   NODE_MODULE_CONTEXT_AWARE(NODE_GYP_MODULE_NAME,                     \
   ^
../src/wpi.cc:7:1: note: in expansion of macro ‘NODE_MODULE_INIT’
 NODE_MODULE_INIT() {
 ^

../src/wpi.cc: In function ‘void _register_wiringPi()’:
/home/pi/.cache/node-gyp/10.16.0/include/node/node.h:579:17: error: redefinition of ‘void _register_wiringPi()’
     NODE_C_CTOR(_register_ ## modname) {                              \
                 ^
/home/pi/.cache/node-gyp/10.16.0/include/node/node.h:562:25: note: in definition of macro ‘NODE_C_CTOR’
   NODE_CTOR_PREFIX void fn(void)
                         ^
/home/pi/.cache/node-gyp/10.16.0/include/node/node.h:607:3: note: in expansion of macro ‘NODE_MODULE_X’
   NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
   ^
../src/addon.h:77:39: note: in expansion of macro ‘NODE_MODULE’
     #define NODE_MODULE_DECLARE(name) NODE_MODULE(name, nodemodule::init)
                                       ^
../src/wpi.cc:25:1: note: in expansion of macro ‘NODE_MODULE_DECLARE’
 NODE_MODULE_DECLARE(wiringPi);
 ^
/home/pi/.cache/node-gyp/10.16.0/include/node/node.h:598:17: note: ‘void _register_wiringPi()’ previously defined here
     NODE_C_CTOR(_register_ ## modname) {                              \
                 ^
/home/pi/.cache/node-gyp/10.16.0/include/node/node.h:562:25: note: in definition of macro ‘NODE_C_CTOR’
   NODE_CTOR_PREFIX void fn(void)
                         ^
/home/pi/.cache/node-gyp/10.16.0/include/node/node.h:611:3: note: in expansion of macro ‘NODE_MODULE_CONTEXT_AWARE_X’
   NODE_MODULE_CONTEXT_AWARE_X(modname, regfunc, NULL, 0)
   ^
/home/pi/.cache/node-gyp/10.16.0/include/node/node.h:634:3: note: in expansion of macro ‘NODE_MODULE_CONTEXT_AWARE’
   NODE_MODULE_CONTEXT_AWARE(NODE_GYP_MODULE_NAME,                     \
   ^
../src/wpi.cc:7:1: note: in expansion of macro ‘NODE_MODULE_INIT’
 NODE_MODULE_INIT() {
 ^
```
Shoutout to possibly related issues:
#94 #90 #75 #74 #73 #80 #89